OpenRTX/OpenRTX #172: Nightly Builds: DM1801 and GD77 missing

marcoSchr, 2023-08-21:
At the moment we build the firmware for the devices when new code is pushed to master.
We only update the wrap files to the nightly folder.
The DM1801 and GD77 do not use the wrap firmware and are only build as binary firmware.
Thus they are never uploaded.

We would need to upload the bin files for those to the nightly folder.
In order to do so I see two possibilities:

1. Upload all bin files
2. Upload the specific bin files for the two targets

The first solution would be simpler to maintain, but would upload the bin firmware files for all other radios as well.
The second solution would need more work to be maintained, but would not upload the additional bin firmware for all the other targets.

silseva, 2023-08-22:
The GD77 and DM1801 have their own wrapping script, outside `radio_tool`, because the encryption system is different. However the `_wrap` build target is defined also for those devices.

Regarding the fact that the binaries are not present in the nightly builds is mostly due to the fact that the development for those two targets is currently dormant, as they cannot support M17 and currently they still not have the persistence of settings. This said, I have nothing against including them in the nightly builds. :)

turnrye, 2023-08-23:
Trying to work this as part of https://github.com/OpenRTX/OpenRTX/issues/173 by making each push include build artifacts, but Im facing issues getting the wrapping script to work. Help wanted.

silseva, 2023-08-24:
What kind of issues do you have?

silseva, 2024-01-13:
Nightly builds of the firmware images for GD77 and DM-1801 are now available as .sgl files. There was an issue on the server preventing them to be uploaded on the web page.

silseva, 2024-01-16:
Nightlies tested on DM-1801, now binary images are ok.

closed 2024-01-16 by silseva