OpenRTX/OpenRTX #175: Target: T-TWR Plus

edgetriggered, 2023-08-23:
### What

The purpose of this issue is to track high level features supported by the OpenRTX port to [LilyGo T-TWR Plus](https://www.lilygo.cc/products/t-twr-plus). Activity for this experimental work is taking place in the [ttwrplus](https://github.com/OpenRTX/OpenRTX/tree/ttwrplus) branch, while SA868S bandband firmware is tracked separately in the [sa8x8-fw](https://github.com/openrtx/sa8x8-fw) repository. Unlike other OpenRTX targets, which are ARM Cortex-M based and supported by Miosix, this platform uses the [Espressif ESP32-S3](https://www.espressif.com/en/products/socs/esp32-s3) system-on-chip with Xtensa LX7 cores.

### Who

The primary contacts for this port are:

- @n1zzo 
- @edgetriggered

Collaboration for this effort is taking place through the [OpenRTX Matrix Space](https://matrix.to/#/#openrtx:matrix.org) and/or [Discord](https://discord.gg/TbR2FVtMya), in the relevant T-TWR Plus room/channel.

### Resources
- [TWR Plus 2.0 Schematics](https://github.com/Xinyuan-LilyGO/T-TWR/blob/master/schematic/T-TWR-Plus_Rev2.0.pdf)
- [Zephyr API Documentation](https://docs.zephyrproject.org/apidoc/latest/index.html)

***

- [X] Preliminary OpenRTX application bootable with [Zephyr RTOS](https://github.com/zephyrproject-rtos/zephyr)
- [X] Power Management Unit supported through [XPowersLib](https://github.com/Xinyuan-LilyGO/T-TWR/tree/master/lib/XPowersLib)
- [ ] Baseband signal routing through analog switches
- [X] FM transmit
- [X] FM receive
- [ ] M17 transmit
- [ ] M17 receive
- [X] GPS receive
- [ ] External microphone + speaker through USB audio interface
- [ ] External microphone + speaker through Bluetooth or WiFi

sp5wwp, 2023-11-22:
Can the MCU transmit arbitrary baseband over FM now?

silseva, 2023-11-24:
Yes, provided that you inject the signal externally, as the output stream driver is not implemented yet. The AT1846S driver can already support the transmission of digital modes.

RE-Specto, 2025-04-02:
Hi! with [T-TWR v2.1](https://lilygo.cc/products/t-twr-rev2-1) released a while ago, I wonder if there's any plans for this small and promising platform? as far as I understand the development was waiting for the v2.1 release so the baseband signal routing through analog switches is available at the hardware level, which afaik now is the reality.

RE-Specto, 2025-04-02:
thank you @silseva 
I really wish to find time.. ([still owe you](https://github.com/OpenRTX/OpenRTX/issues/275) a testing of the UV390 Plus 10w firmware as well :) and also have the v2.1 laying around on the shelf, as well as some [previous experience](https://github.com/RE-Specto/ESP32-NuteBarrels-Controller) with ESP32 (platformio with arduino library imported). but for now I'm massively overworked with my full time job, so it is hard to make any promises. nevertheless, I'm grateful to hear that there are plans for this platform, and hopeful to find time one day to give a helping hand with this one. I really believe that porting it to this platform will open a door to M17 for the masses, as it is finally an affordable off the shelf device to support the protocol, as well as a widely available micro-controller that many alternative custom designs can be delivered with.

RE-Specto, 2026-02-21:
thank you @silseva 
looks like the bot is pretty aggressive with setting the stale tag.