OpenRTX/OpenRTX #179: Festure: Add target for linux with Module17 UI

marcoSchr, 2023-08-25:
As the `Module17` has a different kind of UI than other targets, it will be useful to compile it for linux as will.
This will allow to test changes with the `Module17` UI.

silseva, 2023-08-25:
There is something already set up in the meson.build: https://github.com/OpenRTX/OpenRTX/blob/ec57e74bc6ede3ef2dd67cceb479547003e905d9/meson.build#L292
It also requires to have the 128x64 screen emulation selected a few lines below. We can probably make it a separate target, effectively.

silseva, 2023-09-06:
I saw the errors: the Module17 UI code contains some references to a data structure that is defined only in the Module17 code. I'll see how to provide a stub for it in order to make the code compile also under linux

silseva, 2023-11-05:
Fixed the build process with commits #12e2a0f and #6b34bf3. Is now possible to compile OpenRTX linux with the Module17 UI.

closed 2023-11-05 by silseva