OpenRTX/OpenRTX #184: Module17 - baseband level tuning aid

sp5wwp, 2023-09-07:
A piece of code would provide an additional menu entry showing peak-to-peak voltage value for the incoming signal. That would make having a scope not necessary to tune the input baseband level. The value displayed should be a peak-to-peak value for a time window of about 100-200 ms. For board revisions with digital potentiometers, this could allow for automated RX level tuning - just tune the radio to static and let the algorithm run - set the potentiometer to the highest value that doesn't clip the signal OR apply at least S9 M17 RF signal and run the same procedure (maybe take 90% of the computed setting). The latter method guarantees best dynamic range for a valid signal.
Another piece of code would generate a tone with a precise frequency and amplitude. This tone would be used to tune the frequency deviation using Bessel null method. This process can not be automated and requires user's attention and running a receiver (general purpose SDRs work great here).

Comments are welcome, as always :)

mrickey, 2024-05-17:
I would welcome this, setting levels on the Module17 seems kind of "open loop" to me.

I really like the addition of an analog tone output for setting a bessel null using a spectrum analyzer or SDR receiver.

sp5wwp, 2024-06-09:
More developers are needed to implement this. M17/OpenRTX projects are heavily under-resourced.