OpenRTX/OpenRTX #188: Discussion: M17: Storing Callsign Data

marcoSchr, 2023-09-16:
# Current behavior

- The source callsign is saved in `status->M17_src`
- The destination callsign is saved in `status->M17_dst`

If there is extended callsign data:
- The first callsign is saved in `status->M17_orig`
- The second callsign is saved in `status->M17_refl`

# Proposed new behavior

If there is no extended callsign data:
- The source callsign is saved in `status->M17_src`
- The destination callsign is saved in `status->M17_dst`

If there is extended callsign data:
- The source callsign is saved in `status->M17_link`
- The destination callsign is saved in `status->M17_dst`
- The first callsign is saved in `status->M17_src`
- The second callsign is saved in `status->M17_refl`

# Why the new behavior

Assuming the User is most interested in the original source and the destination.
We would need to look only at `status->M17_src` and `status->M17_dst`.
This will allow the UI code to be simpler.
This is achieved, as there is no need to check for extended callsign data when displaying source and destination callsign.
If extended callsign data is available, the corresponding data can be shown without the need to switch anything.
This will also allow to optionally not display the additional information, for example if the user does not want to see it or if the display is to small.

marcoSchr, 2023-09-26:
This change will also be usefull for #195, as no additional code would be needed there.

silseva, 2023-09-27:
Given that I have no means of testing M17 traffic over repeater/hotspot/reflector: you are saying that in these cases the first callsign field in the metadata (referred to as "originator" in the M17 specification) contains the callsign of who initiated the transmission. Did I understand?

I.e. supposing that we have AN7ANI transmitting via the repeater AA2BBB to all on reflector M17-TEST A, I should have the following:
- `src` = AA2BBB
- `originator` = AN7ANI
- `refl` =  M17-TEST A
- `dst' = BROADCAST

marcoSchr, 2023-09-27:
Yes, this is true.
And I want to store the variables locally swapped.
So that `M17_src` in this case would contain the`originator` callsign.
And `M17_orig` would be renamed and store the callsign of the hotspot/repeater.

This will allow us to swap them once on receive.
We then do not need to specially check for them in the UI code in multiple positions.

silseva, 2023-10-14:
I think we can close this issue, the proposal has been applied to the code.

closed 2023-10-14 by marcoSchr