OpenRTX/OpenRTX #197: Feature: add T-TWR Plus target into nightly build

lyusupov, 2023-10-05:
Is there any chance to add the T-TWR Plus target:
https://twitter.com/OpenRtx/status/1704954487184453848
into the [nightly build](https://files.openrtx.org/nightly/) ?

silseva, 2023-10-05:
Is already in the plans, we'll do it soon!

lyusupov, 2023-10-05:
Thank you for the quick response!

Please, be aware that **T-TWR Plus** board  _(both current V2.0 and upcoming V2.1)_ is already supported by Adafruit TinyUF2 bootloader:  https://github.com/adafruit/tinyuf2/tree/master/ports/espressif

[image]

Further batches of the product may come with the bootloader been pre-installed on the factory.

Thus, It could make sense to deploy not only the bare .bin binaries,
but the UF2 one as well.

Converting an ESP32-S3 'application' binary file info UF2 format is as simple as:
```
uf2conv.py firmware.bin -c -f 0xc47e5767 -b 0 -o firmware.uf2
```

Please, read https://github.com/microsoft/uf2 for more details.

n1zzo, 2023-10-17:
Taking a look at this, planning to release v3.7 with T-TWR Plus builds in UF2 format.