OpenRTX/OpenRTX #198: Firmware issues with new RT3s models

silseva, 2023-10-08:
DB2OE and ON4MOD reported the following issues affecting OpenRTX when flashed on new Retevis RT3s radios:

1. settings not persistent, except the first time the radio is powered up after a firmware update
2. missing TX in FM mode
3. firmware freezing when switching to M17 mode 

The first problem may be related to the flash requiring a full sector erase before each write.

silseva, 2023-10-08:
The discussion here provides some hits, as the symptoms are similar: https://www.opengd77.com/viewtopic.php?t=2667
The MCUs installed in the new RT3s may be APM32F405: https://global.geehy.com/apm32?id=49

silseva, 2023-10-08:
Test firmware to try tackle point 3: hw version field of info page contains the DBGMCU_IDCODE, interrupt level of audio stream drivers raised from 10 to 7.

[openrtx_mduv3x0_test.zip](https://github.com/OpenRTX/OpenRTX/files/12841594/openrtx_mduv3x0_test.zip)

UnImAtrIXzEr0, 2023-10-10:
I now believe there is a problem with radio_tool under Linux. As soon as I write with the manufacturer's flash tool under Windows 10, the device with M17 does not freeze. FM transmission also works perfectly. That's why we suspect that the radio_tool and the manufacturer's software flash slightly differently.

So I don't think errors 2 and 3 are a firmware problem.

But point 1 remains, saving doesn't work properly yet.

silseva, 2023-10-10:
This is very interesting. Just to have another test, can you flash the latest nightly build with both the OEM firmware update tool and `radio_tool`, please?
We'll also organize to buy a new UV380/RT3s in order to have a sample of the latest hardware.

UnImAtrIXzEr0, 2023-10-11:
Ok, I installed the nightly openrtx_mduv3x0_wrap from october 8th with radio_tool. 
Now the radio works just before flashed with OEM firmware update tool. The strange thing is that after flashing the settings are still there, such as callsign and frequency.
Good up to this point.

silseva, 2023-10-22:
After some investigations with Morgan ON4MOD we found out that there is something weird going on when writing to the MCU flash memory - were we're storing the settings and persistence now - and it also seems that the new MD-UV380/RT3s have an APM32F4. From the tests we run, it seems that only the first byte gets written to the flash and then the memory goes into an unknown state were it does not write anything else. The only possible solution to this issue, right now, is to do a full sector erase before each write, which is something that reduces the lifespan of the flash memory.

Given that soon we'll have codeplug support on the external flash and that we'll also move the persistence there, I'll prefer to invest time on that rather than trying to find a solution for the non-persistent settings issue, also because I don't have a recent RT3s.

iz4tow, 2023-11-27:
so? can you provide the istructions? Thanks

silseva, 2025-12-02:
Thanks for the research!
Actually I prefer to have a solution based on runtime detection of the MCU type, hoping that the content of the CPUID register is different between the STM32 and APM32 devices...
The thing is that we're going towards a solution using the external flash memory to store the device settings and persistent state, so MCU flash will be abandoned at some point. However, given that I cannot predict how much time this migration work is going to take, having the current driver fixed is also good. Do you want to take care of it?