OpenRTX/OpenRTX #199: T-TWR Plus fails to complete the boot sequence when SA868 uses NiceRF firmware

lyusupov, 2023-10-19:
Device: LilyGO T-TWR plus V2.0 VHF
Bootloader: TinyUF2 
Firmware: 10/19/2023 nightly build

This is the boot log:

```
*** Booting Zephyr OS build zephyr-v3.4.0-3837-ga4de2eb3d175 ***
->  getSysPowerDownVoltage:2600
DCDC=======================================================================
DC1  : +   Voltage:3300 mV
DC2  : -   Voltage:900 mV
DC3  : -   Voltage:3400 mV
DC4  : -   Voltage:1100 mV
DC5  : +   Voltage:3300 mV
ALDO=======================================================================
ALDO1: +   Voltage:3300 mV
ALDO2: +   Voltage:3300 mV
ALDO3: +   Voltage:3300 mV
ALDO4: -   Voltage:3300 mV
BLDO=======================================================================
BLDO1: +   Voltage:3300 mV
BLDO2: +   Voltage:3300 mV
===========================================================================
PowerKeyPressOffTime:10 Second
PowerKeyPressOnTime:128 Ms
===========================================================================
Val = 22
Setting Charge Target Current : 797
Setting Charge Target Voltage : 4000
SA8x8: baseband is not ready!
SA8x8: baseband is not ready!
```

The sequence does not proceed beyond this point.

It would be nice if the boot could proceed all the way down to full featured OpenRTX OLED UI.
Then the UI can indicate that RF module is missing or has wrong firmware version.

silseva, 2023-10-20:
Hi!
We currently do not support the original SA868 firmware, this is why it fails to detect it. In the future we'll show an error message on the screen to inform the user about this.

lyusupov, 2023-10-20:
@silseva 
> We currently do not support the original SA868 firmware

It's Ok.
I mean that wrong or missing RF module should not stop the boot sequence.

Looks like we will stay in the loop all the time when there is no 'OK' response or baud rate of the poll is different from the baud rate of a module.
https://github.com/OpenRTX/OpenRTX/blob/master/platform/drivers/baseband/SA8x8.c#L98-L112
```
static inline void waitUntilReady()
{
    char buf[SA8X8_MSG_SIZE] = { 0 };

    while(true)
    {
        uartPrint("AT\r\n");
        int ret = k_msgq_get(&uart_msgq, buf, K_MSEC(250));
        if(ret != 0)
            printk("SA8x8: baseband is not ready!\n");

        if(strncmp(buf, "OK\r", SA8X8_MSG_SIZE) == 0)
            break;
    }
}
```
Should it be a timed loop instead ?

silseva, 2023-10-20:
Sure, it has to be changed to a timed loop. I did it yesterday but then the radio seems to hang somewhere else, I have to investigate more.

lewisxhe, 2023-11-14:
@lyusupov The version you have also supports SA868 firmware written to openRTX

lyusupov, 2023-11-14:
@lewisxhe 
Sure! But I am not ready to erase genuine SA969  firmware right now.
I still need it until OpenRTX one will support the same functions that genuine firmware has.