OpenRTX/OpenRTX #209: M17: Noise when not receiving anything

marcoSchr, 2023-11-17:
# Observed behavior

When the radio is in M17 mode and not receiving M17, the speaker outputs noise.

# Expected behavior

The speaker is silent when not receiving anything

# Info

- hardware: RT3S

silseva, 2023-11-17:
Hi!
Do you mean white/electric noise? Is because when there is no speech the speaker amplifier is kept active and picks up some electrical noise, I think mainly from the screen. There is space for improvements in how the audio is managed for the M17 operating mode.

marcoSchr, 2023-11-18:
Here is a video of the issue.
There is some background noise, but you can clearly hear the noise from the device when first turned on and at the 41 second mark, when switched from FM to M17. As you can hear, there is no noise when in FM mode.

https://github.com/OpenRTX/OpenRTX/assets/49691247/634e8009-3aad-4927-b46b-93daefbff1ab

silseva, 2023-11-24:
The noise you hear is due to the fact that in M17 mode the audio path towards the speaker is opened when the operating mode is selected and never closed until the mode is changed. The effect of this is that the AF amplifier remains active, making the speaker outputting the noise picked up on the PCB traces.
As a future improvement, we should add the possibility to mute an audio path, that is keeping it active but with the destination endpoint disconnected. This will allow to keep the codec2 thread running while turning off the audio amplifier when there is no M17 activity.

silseva, 2024-02-06:
Issue solved.

closed 2024-02-06 by silseva