OpenRTX/OpenRTX #221: "For emergency use press any button" screen appears after flashing.

jasiek, 2023-12-27:
Hey,

I've flashed [v0.3.6](https://github.com/OpenRTX/OpenRTX/releases/download/v0.3.6/openrtx_gd77_v0.3.6.sgl) on a Radiooddity GD-77 device with an existing codeplug - when I power up the radio I see this screen:

[image]

This screen can be bypassed by pressing any button within the first second of the screen appearing. When accessing the battery info screen I see that the voltage gradually grows to 8.1V and battery charge grows to 100% after ~20 seconds.

If this is a problem assessing the battery level accurately (this radio has been in storage for about a year), then perhaps disabling this feature would be a good idea.

jasiek, 2023-12-27:
The firmware has been flashed using `radio_tool`.

silseva, 2023-12-27:
It seems to be an issue with the measurement of battery voltage at startup, with the ADC returning 0V on the first acquisition. Successive readings are ok and thus you see the voltage going up.

jasiek, 2023-12-27:
Indeed, there is a problem that this screen seems to lock up the radio. If I don't press a button within the first second, the radio becomes frozen.

silseva, 2024-01-09:
Also the DM-1801 is affected. It seems that these radios have an hardware low-pass filter on the battery voltage probe, which causes the measured voltage to increase a bit slowly since the power up. The digital low-pass filter in the state update function further exacerbates this effect causing the low battery alarm to kick in unconditionally.

Problem has been fixed with commit  c4b7b7a

closed 2024-01-09 by silseva