OpenRTX/OpenRTX #229: Module17 no audio for DST ECHO and INFO

rolf-tx, 2024-01-24:
I have a Module17 v0.1e (Firmware nightly Feb23 v0.3.6-50) used FTM300 and WPSD on a RPI2W.
All settings default except callsign.

When setting target to ECHO on mod17 the transmission to WPSD works ok but mod17 does not play audio when receiving data with DST ECHO.
Likewise if setting target INFO the audio is not played for data received with DST INFO.

I would expect ECHO and INFO audio to be played.

I verified with the windows M17-RTX app that WPSD did indeed return audio for INFO and ECHO targets. Also works when initiated from mod17.

I suspect mod17 is filtering out anything except target ALL and mod17 callsign.

rolf-tx, 2024-01-24:
I did a quick hack in OpMNod_M17.cpp to add INFO and ECHO and now I can get ECHO and INFO.
Can you add some setting option to enable ECHO and INFO?

bool OpMode_M17::compareCallsigns(const std::string& localCs,
                                  const std::string& incomingCs)
{
    if(incomingCs == "ALL" **|| incomingCs == "INFO" || incomingCs == "ECHO"**  )
        return true;

silseva, 2024-01-24:
I think we can just apply your change and solve the problem. Adding a dedicated setting entry is overkill, in my opinion.

commit 3f0cfe94 (2024-01-28): M17: always open squelch on tramsissions with destination set to INFO or ECHO

Fix #229
closed 2024-01-28 by silseva