OpenRTX/OpenRTX #230: Module17 v0.1e fails to shutdown - off pulse too short

rolf-tx, 2024-01-24:
Module17 v0.1e fails to shutdown when powered by barrel connector because the off pulse on POWER_SW pin is too short and firmware gets to execute platform_init restoring POWER_SW on before the power supply has collapsed and MCU shutdown.

Suggested fix:
Add 1 sec delay in platform_terminate function in platform/targets/Module17/platform.c after gpio_clearPin(POWER_SW) 

```
void platform_terminate()
{
    /* Shut down LEDs */
    gpio_clearPin(PTT_LED);
    gpio_clearPin(SYNC_LED);
    gpio_clearPin(ERR_LED);

    adc1_terminate();
    nvm_terminate();
    audio_terminate();

    gpio_clearPin(POWER_SW);
    /* Wait 1s to allow power supply to shutdown properly*/
    sleepFor(0, 1000); 
}
```

silseva, 2024-01-24:
This is interesting. We thought that the main reason for this was an hardware bug due to a missing pull-down resistor on the base of the transistor controlling the P-Mos acting as power gate (refer to Module17 schematics on M17 github). I'll try your fix on my board in the next days and then integrate it into mainline.

rolf-tx, 2024-01-24:
I believe the schematic is correct, a PMOS requires a pull-up to turn off.
Assuming VIN=12V, Q4 source=12V.:
Power on: Assume Q5 is off, the Gate of Q4 is pulled high to 12V by R66 making Vgs=0V, and turning off the PMOS.
Power off: Assume Q5 is on, the Gate of Q4 is pulled 'low' to 6V by R66/R67 divider, making Vgs=-6V, and turning on the PMOS.

silseva, 2024-01-26:
[image]

This is the voltage measured on the NPN base: as you can see it briefly go down to 0V when the MCU pulls down the line and then rises to 1.3V when the mcu shuts down and its gpios go to Hi-Z.
I'll add the delay, as you suggested, and see how it goes.

rolf-tx, 2024-01-27:
Interesting, I would not have expected the hiz IO to provide that much current, it would need to be 13uA just for the 100k.
I believe your plot is actually the GPIO voltage or just after the schottky.

I looked closer at GPIO_POWER. It is low for about 5us before the program issues the reset to the MCU and it enters highz mode.
I agree a pull down on GPIO_POWER would most likely allow Q5-base to fully discharge during post reset to POWER_SW on time and system power down. I would still recommend adding a delay.

**Summary:**
The main cause seems to be that 
Q5-base doesn't get pulled fully to 0V before the program resets the MPU. I believe due to high value of 100k pull down.
Thus 12V power-down is never started.

**With the delay add**
Q5-Gate is fully discharged and 12V power down is started.
The MCU internal power monitor will assert Power Down Reset at VDD<1.68V.
The reset would be deasserted at VDD>1.72 or >2.19V if brown out reset is enabled.
Also observed GPIO_POWER never goes to 1.3V hiz output state, thus never turning Q5 back on and system remains powered off.

**Some Plots of without delay added**
Q5-Base (Yellow) only gets pulled down to 0.32V.
Q4-Gate only gets pulled up to about 7V.
[image]

12V output it does not actually have time to decay at all.
[image]

**Plots with the delay inserted after POWER_SW low.**
Q5 base (yellow) now stays low. 
Q4 Gate it now gets all the way to 12v. 
[image]

The 12V rail (pink) can be seen to decay to 5V in 1ms.
[image]

The 3.3V VDD rail (pink) collapses to ~1.8V in 1.5ms.
[image]

And 3.3V finally collapses in to 0V in about 50ms.
[image]

GPIO_POWER (Pink) is never goes to 1.3V state.
[image]

silseva, 2024-01-28:
Yes, sorry: my plot is the voltage on pin 3 of D7.
Anyway, many thanks for your measurements! I'll add the delay at the end of `platform_terminate`, as you suggested.

commit 202a1994 (2024-02-06): Module17: added a 100ms sleep after power off to prevent the board from rebooting.

Fix #230.
closed 2024-02-06 by silseva