OpenRTX/OpenRTX #234: T-TWR Plus builds failing

edgetriggered, 2024-02-01:
The current meson build definition includes logic that uses system-installed codec2 unless performing cross builds. Zephyr targets like T-TWR Plus do not use meson's cross build mechanism but do need codec2 cross compiled from the subproject:

```
% rm -rf build; meson setup build; meson compile -C build openrtx_ttwrplus_uf2
The Meson build system
Version: 1.3.1
Source dir: /tmp/openrtx-build/OpenRTX
Build dir: /tmp/openrtx-build/OpenRTX/build
Build type: native build
Project name: OpenRTX
Project version: 0.3.6
C compiler for the host machine: ccache cc (gcc 13.2.1 "cc (GCC) 13.2.1 20230801")
C linker for the host machine: cc ld.bfd 2.41.0
C++ compiler for the host machine: ccache c++ (gcc 13.2.1 "c++ (GCC) 13.2.1 20230801")
C++ linker for the host machine: c++ ld.bfd 2.41.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library codec2 found: YES
...
-- Configuring done (1.9s)
CMake Error at CMakeLists.txt:88 (target_sources):
  Cannot find source file:

    subprojects/codec2/src/dump.c
```

I propose a change similar to the following which allows the build system to clone and use the codec2 subproject rather than the system-installed one for Zephyr targets.

```diff
diff --git a/meson.build b/meson.build
index d84da94..ca89e73 100644
--- a/meson.build
+++ b/meson.build
@@ -119,7 +119,7 @@ openrtx_inc += ['lib/minmea/include']
 openrtx_inc += ['lib/qdec/include']
 
 # CODEC2, open source speech codec, compile from source on embedded
-if meson.is_cross_build()
+if meson.is_cross_build() or get_option('zephyr') == true
   codec2_proj = subproject('codec2')
   codec2_dep  = codec2_proj.get_variable('codec2_dep')
 else
@@ -614,6 +614,7 @@ ttwrplus_opts = {
   'c_args'             : [],
   'cpp_args'           : [],
   'link_args'          : [],
+  'zephyr'             : true
 }
 
 ##
diff --git a/meson_options.txt b/meson_options.txt
index 9d797cb..2d04589 100644
--- a/meson_options.txt
+++ b/meson_options.txt
@@ -1,3 +1,4 @@
 option('asan', type : 'boolean', value : false, description : 'Compile the software with AddressSanitizer')
 option('ubsan', type : 'boolean', value : false, description : 'Compile the software with Undefined Behaviour Sanitizer')
 option('test', type: 'string', description: 'Replace the main OpenRTX source file with a specialized test')
+option('zephyr', type: 'boolean', value : false, description : 'Target uses Zephyr RTOS')
```

silseva, 2024-02-02:
Another alternative is to remove the `if meson.is_cross_build()` and always download the codec2 sources

n1zzo, 2024-02-05:
This works, if you agree I'll commit this!

```
[nizzo@highlander OpenRTX]$ git diff
diff --git a/meson.build b/meson.build
index d84da943..2d072469 100644
--- a/meson.build
+++ b/meson.build
@@ -119,8 +119,8 @@ openrtx_inc += ['lib/minmea/include']
 openrtx_inc += ['lib/qdec/include']

 # CODEC2, open source speech codec, compile from source on embedded
+codec2_proj = subproject('codec2')
 if meson.is_cross_build()
-  codec2_proj = subproject('codec2')
   codec2_dep  = codec2_proj.get_variable('codec2_dep')
 else
   cpp        = meson.get_compiler('cpp')
   ```

edgetriggered, 2024-02-05:
This solution is fine. If the emulator is used more commonly for codec experimentation, then it may be best to just always build the codec from source.

n1zzo, 2024-02-09:
Fixed by 7aa9385ec3be2a91d3e8c9161f8a260fa7286864.

closed 2024-02-09 by n1zzo