OpenRTX/OpenRTX #241: GPS Display broken after switching to sniprintf

marcoSchr, 2024-02-18:
# Observed behavior 

When the GPS gets a fix, the display does not show the current position

# Expected behavior

When the GPS gets a fix, the display does show the current position

# Steps to reproduce

1. Enable GPS
2. Go to GPS screen
3. Wait for fix

# Additional Information

I have tested this on a Retevis RT3S, but I assume it is broken on all devices with GPS support.
I have used git bisect to find the commit, which introduced the bug: 1791e70eef0126c227519b0fec2749635b02ae65.

[image]

silseva, 2024-02-18:
The function to be fixed is this one: https://github.com/OpenRTX/OpenRTX/blob/ce8e7583480c786aa8a429f9f93e40f8593fdc6d/openrtx/src/ui/default/ui_menu.c#L632
The `iprintf` function is a stripped-down `printf` without floating point support, which means that the `%f` specifiers do not get parsed correctly. The fix should be fairly easy and consists of using a `%d.%d` notation as in the VFO frequency display.

marcoSchr, 2024-02-18:
I think we can improve this further by using the same format internally for storing GPS data as the proposed [codeplug](https://github.com/OpenRTX/openrtx.github.io/blob/master/binary_cps_format.md#geo_t-type-description).
There the coordinates are stored as an 8bit integer decimal part and a 16 bit integer fractional part.

The minnmea library stores the coordinates as integers internally and we use a function to convert it to a float:
https://github.com/OpenRTX/OpenRTX/blob/ce8e7583480c786aa8a429f9f93e40f8593fdc6d/lib/minmea/include/minmea.h#L246

https://github.com/OpenRTX/OpenRTX/blob/ce8e7583480c786aa8a429f9f93e40f8593fdc6d/openrtx/src/core/gps.c#L93

silseva, 2024-02-19:
Yes, switching to a format using only integer values will be a good improvement since it will make the code more easily portable also to devices without an FPU (otherwise you'll need to bring in the softfp libraries, which are heavy). The `geo_t` structure of the codeplug works fine, the only modification to do is to use an `int16_t` value for the altitude and drop the +500m offset. A signed 16 bit integer allows to store an altitude between +32768m and -32768m, which is good enough (unless we'll put the code on a strato balloon, maybe).

marcoSchr, 2024-02-20:
Using a signed 16 bit integer for altitude sounds good. We would only loose the ability to display fractional altitude. I think showing only whole meters is fine.

silseva, 2024-02-27:
A resolution of +/- 1m should be fine, also considering that the average position error of a GPS in in the order of the meters.

silseva, 2024-04-13:
Fixed, thanks Marco for the help!

PR #258 (2024-04-05): Change how gps data is stored and fix gps display
This will fix the display of the GPS data.

It changes the internal data types of the GPS data.
It adds a test for the conversion function.

This will fix #241
closed 2024-04-13 by silseva