OpenRTX/OpenRTX #245: Battery percentage display does not feel acurate

marcoSchr, 2024-02-18:
At the moment OpenRTX assumes, the battery voltage drops linearly between 100 and 0 percent.

In order to find an approximation of the battery percentage, we would need to measure a battery from full to empty in relation to how the voltage changes over time. With this we could get a better approximation of battery time left.

silseva, 2024-03-02:
One issue is an approximate value of the voltage divider gain set in the `platform_getVbat` functions, which has to be corrected for all the radios.
More in general, at least the TYT firmware stores the thresholds for "battery empty" and "battery full" voltages in the calibration data. From those, it computes the four intermediate steps used to show the bars in the battery icon. Internally, the firmware uses the raw ADC value of the battery voltage to determine the charge level and the thresholds are computed by taking the data from the calibration and shifting them left by four (they are 8-bit values). We have to see if is possible to use the values from the calibration, at least for the radios having them.

silseva, 2024-11-10:
Commit df5d6dc increases the accuracy of ADC measurements by rescaling them to the actual Vref. This should improve the estimation of the battery state of charge.