OpenRTX/OpenRTX #248: Add an option to invert PTT logic on Module17

ManoDaSilva, 2024-02-25:
Hey! In order to support the IC-F5060, we had to invert the logic in both PTT in and out by manually change the logic as it's active high... 

Is it possible to add an option on the Module17 config menu to invert that PTT logic?

silseva, 2024-02-27:
Yes, I'll add this as soon as I have some time

silseva, 2024-03-06:
Do you have suggestions for the name of the menu entry and the valued displayed?
I started with a "PTT level" entry label with "high" and "low" as possible values, but I'm not completely sure is understandable.

ManoDaSilva, 2024-03-06:
Perhaps "PTT Logic In" and "PTT Logic Out" labels, if there's enough space?

silseva, 2024-03-07:
I'd go for "PTT in" and "PTT out" la else, with "Act Hi" and "Act Lo" as values.

silseva, 2024-03-13:
Done!

closed 2024-03-13 by silseva