OpenRTX/OpenRTX #263: MD-390G Frozen M17 freq entry

KC5CQW, 2024-04-12:
I have a newly (today) modded MD-390G UHF that locks up when entering a frequency in M17 mode. Specifically, my hotspot at 432.55MHz. FM has good RX/TX into a local repeater. I nuked my HS bootloader so it isn't working right now to test DMR/M17 further.

silseva, 2024-04-13:
Which firmware image did you flash? And with which tool?
In the past we found out that some of the TYT firmware update tools seem not to write the entire firmware image in the MCU flash, and this causes lockups/crashes.

KC5CQW, 2024-04-14:
Good info, I used an old TYT program. I didn’t know of any alternatives. 73, Damon

silseva, 2024-04-14:
I need to update the README and the flashing instructions.
Let me know if you solved! :)

silseva, 2024-06-21:
I assume you solved the problem, if not feel free to reopen the issue.

closed 2024-06-21 by silseva