OpenRTX/OpenRTX #266: [MD380] Weak RX Audio level on M17

HB9HLI, 2024-04-26:
Hello,
I modified an MD380 UHF (not UV) and installed the latest firmware release. On FM, everything works fine.
On M17, the RX audio is very weak and I have to turn the volume up to the maximum to hear a very weak sound.
Could this be due to an error in the hardware modification of the MD380 or something that can be improved by the firmware? 
On transmit the levels and audio quality are excellent.

Thanks and best 73's, Dom. HB9HLI

silseva, 2024-04-27:
Hi Dom,
your mods are working fine, the problem is on the firmware side. More in detail, now we're using a filtered PWM signal to generate the demodulated audio, but the analog path from the PWM filter to the speaker introduces a good amount of attenuation. There is an alternative path which uses the HR_C5000 but I didn't worked on it yet because of lack of free time. In any case I hope to work on it soon.
You can also try this mod https://github.com/travisgoodspeed/md380tools/issues/98

ddibbets, 2025-01-11:
Just a feedback on this.
The firmware works well on the MD-UV380 - volume works and its loud.
On the MD-380 I tried the mod above (https://github.com/travisgoodspeed/md380tools/issues/98); the good thing is the volume works - the bad is that at its max its not louder than before the mod.  @silseva did you have any time to work on the alternative path using the HR_C5000?

ddibbets, 2025-01-11:
Yes, tried today with latest nightly.

onemikedelta, 2025-04-09:
My radio (UHF MD390 GPS) also suffers low audio problem. I tried the mod (https://github.com/travisgoodspeed/md380tools/issues/98) however nothing changed. I tried to decrease resistor value down to 1k step by step but change is barely noticeable even on 1k. I traced the line and tried to get signal before the 22k resistor which is near to mcu, it become usable however, all of these trials caused distorted or unmodulated rf signal. I believe modulation signal is produced by the same pin.

Now, I am thinking about using an analog switch ic and isolate the signals according to rx/tx state but I don't wanna be too invasive at this point. Second option is adding an opamp in between R341 and R253 and apply some gain.

My PCB version is V4.1-160322

Is there anything else I can try? Will HR_C5000 be able to produce this signal in the near future?