OpenRTX/OpenRTX #275: MD-UV390 Plus 10W support

RE-Specto, 2024-06-30:
Hi!
recently, there's a new model of the UV380 appeared on the market, the MD-UV390 Plus. it have two versions, 5W and 10W. according to [opengd77 forum topic](https://www.opengd77.com/viewtopic.php?f=19&t=3607) the 5W model is identical in terms of firmware, however the 10W have slight differences in backlight control and a few other alterations. for openGD77 currently there are two versions of the firmware, one for original uv390/plus5w and one for 390plus10w with the changes needed for that hardware. since this new model is quite popular recently, and as it appears the changes required are quite small, I wonder if any OpenRTX support is expected for this model?

silseva, 2024-07-03:
Hi!
I had an exchange of emails with Roger VK3KYY about that radio and he told me about the hardware differences between the two. Supporting also that radio should not be a difficult task, and we can schedule setting up a bsp in the future. Unfortunately at the moment I don't have that radio and I'm a bit busy on other parts of the firmware. In case, do you want to help us with it?

RE-Specto, 2024-07-06:
thanks @silseva   
glad to hear that you have collaborations between you and Roger! its a beauty of open source in action, when projects cooperate rather than competing 👍   
as for helping with this one, I"m indeed interested to take part! testing is something I can definitely do (SQE @ Red Hat :)   
as for the bsp, what kind of work is required?

on a side note, the 390plus that I had access to, is of a friend of mine, so while I can definitely borrow, I cannot modify it. I was about to buy myself one, but haven't ordered yet bc I was waiting for this thread. since you mentioned it's possible, I'll gonna buy it, but it'll definitely take some time for me to obtain for a few reasons.

silseva, 2024-07-10:
The BSP for a radio is composed of a folder in the platform/targets subdirectory and a couple of entries in the meson.build.
In this particular case, given that the plus version is completely equal to an MD-UV380 except for some pin mappings, the work required is to set up the correct pinmap.h and add an entry in the meson.build. I would just keep everything in the target folder for the UV380

RE-Specto, 2024-07-11:
thanks @silseva !  
I'll take a look once I"m back from traveling in a few weeks. hopefully will be able to obtain the radio as well by then.

silseva, 2024-07-18:
Thanks to you!
I dug into my past emails and I found the difference between the "vanilla" MD-UV380 and the PLUS version: they swapped the functionalities of pins for the LCD backlight and RF PA band select. That is, the LCD pin now controls the PA band select and vice-versa.

marcoSchr, 2024-12-24:
I have added a target for the plus version in #300.

RE-Specto, 2025-01-01:
@marcoSchr oh thanks!
@silseva yeah sry life happened so I wasn't been able to take care of it so far (it was in my todo though)

smeegle5000, 2025-02-03:
i have attempted to use the plus branch of the firmware on two md-uv390's, and the rx functionality does not work on m17 or FM, i have confirmed tx to be working fine, i can attach images of the hardware mod if needed, has anyone had any success with this handset?

silseva, 2025-02-03:
The problem may be in the mod, actually. Can you send a picture, please?

smeegle5000, 2025-02-04:
performed on two radios of the same model, resistor is 3.3k with line 332 of audio_codec.c removed on reccomendation from a member in the discord

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

silseva, 2025-02-05:
Please check that the mod wire going from the HR_C6000 baseband input to the MCU is not shorted to ground. Usually this is the most common point of failure when dealing with hardware modifications for M17 and it leads to no RX audio in both FM and M17 modes.

smeegle5000, 2025-02-06:
> Please check that the mod wire going from the HR_C6000 baseband input to the MCU is not shorted to ground. Usually this is the most common point of failure when dealing with hardware modifications for M17 and it leads to no RX audio in both FM and M17 modes.

measured at 1.883 and 1.850 megaohm

marcoSchr, 2025-02-08:
I just went through the available source code of the OpenGD77 project and they have the following differences for the 10W version:

- swapped PA and backlight pins
- different USB product string
- different power levels
- different interpretation of calibration data
- different "calibration data" for ADCs
- option to force 10W

mbalmer, 2025-02-22:
In brussels, silvano stated that he could not tell the two models apart in software, but could the usb product string be used for that?

silseva, 2025-02-23:
It can be a possibility, however we have to determine if the USB product string is stored also in the bootloader firmware

BohdanK-W32, 2025-07-02:
Hi guys!
Can I help with something?

Because now it seems stuck but for me (and I think probably for a lot more people) it's a long awaited device to support

romanatrn, 2025-09-02:
it seems new plus models have gone back to using non plus pin layout for display i tested master and uv3x0p branch and radio backlight only works with master branch.