OpenRTX/OpenRTX #277: [MD-UV380] Fw v0.3.7 can't connect to WPSD

rolf-tx, 2024-07-14:
The UV380 with fw v0.3.7 is unable to connect to WPSD hotspot, no response.
The UV380 can connect to WPSD fine with v0.3.6.

The WPSD behaves like when no valid call sign has been set.
I get the same behavior using v0.3.6.0 if I have a blank call sign.

Module17 can decode the UV380 v0.3.7 ok and looks like it receiving my call sign.
UV380 v0.3.7 can decode WSPD ok.

silseva, 2024-07-15:
Hi!
It seems that something got messed up between the two releases, then. I didn't get the bug because, unfortunately, I don't have an hotspot: all the tests I did were between radios and they worked fine, just like in your case between the radio and the Module17. I'll go through the commits to see if I can get any clue.

marcoSchr, 2024-07-15:
I was able to reproduce this. I bisected the issue to one of two commits:

The first bad commit could be any of:
e53e110a3b8d71355ded8e6af5a6622c07da76b0
8842b98a5243b79f9527439e13ad142e5c45c3fa

marcoSchr, 2024-07-15:
The RT3/MD380 has the same behavior

silseva, 2024-07-16:
> I was able to reproduce this. I see no log output for the MMDVM log on wpsd. The LED for M17 illuminates, but flickers on the MMDVM.
> 
The flickering led may be the symptom of a bad transmission. I'll check with the spectrum analyser.
Can you set the MMDVM log level to verbose/debug and see if it says something, please?

marcoSchr, 2024-07-16:
Here are the MMDVMHost logs.
Once for the failing commit and once for the working commit before.

<details> 
  <summary>8842b98a5243b79f9527439e13ad142e5c45c3fa</summary>

```
D: 2024-07-16 10:02:53.530 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.570 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.610 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.650 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.690 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.775 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.875 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.915 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.955 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:53.995 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.035 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.118 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.157 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.197 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.237 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.277 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.317 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.397 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.437 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.477 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.517 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.564 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.603 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.643 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.683 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.723 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.805 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.845 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.885 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:54.996 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.036 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.076 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.116 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.156 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.196 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.269 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.309 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.350 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.432 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.472 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.512 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.599 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.639 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.679 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.802 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.847 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.882 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.922 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:55.962 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.002 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.042 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.082 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.132 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.172 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.212 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.252 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.292 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.332 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.400 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.440 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.480 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.520 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:02:56.560 M17, raw RSSI: 47, reported RSSI: -47 dBm
[... 172 lines cut ...]
D: 2024-07-16 10:04:17.046 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.046 M17, audio: FN: 87, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.086 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.086 M17, audio: FN: 88, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.126 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.126 M17, LSF text data fragment
D: 2024-07-16 10:04:17.126 0000:  00 00 00 00 00 00 00 00 00 00 00 00 00 00          *..............*
D: 2024-07-16 10:04:17.127 M17, audio: FN: 89, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.166 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.166 M17, audio: FN: 90, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.206 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.206 M17, audio: FN: 91, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.246 M17, raw RSSI: 47, reported RSSI: -47 dBm
[... cut ...]
D: 2024-07-16 10:04:17.326 M17, audio: FN: 94, errs: 0/272 (0.0%)
D: 2024-07-16 10:04:17.366 M17, raw RSSI: 47, reported RSSI: -47 dBm
D: 2024-07-16 10:04:17.366 M17, LSF text data fragment
D: 2024-07-16 10:04:17.366 0000:  00 00 00 00 00 00 00 00 00 00 00 00 00 00          *..............*
D: 2024-07-16 10:04:17.366 M17, audio: FN: 32863, errs: 0/272 (0.0%)
M: 2024-07-16 10:04:17.372 M17, received RF end of transmission from DM4RCO to ALL, 2.9 seconds, BER: 0.0%, RSSI: -47/-47/-47 dBm
```
</details>

silseva, 2024-07-17:
Fixed by 031eda1.
The gpio of the tone generator had its pull-up enabled due to a bug introduced during the refactoring of the gpio driver. This caused the PLL frequency to be shifted of ~2kHz every time the PTT was pressed. A new binary image will be available tomorrow once the nightly build system does its run.

rolf-tx, 2024-07-17:
@silseva 
I compiled it from git master, confirmed working with WPSD.
Thank you for quick fix :-)

closed 2024-07-17 by rolf-tx