OpenRTX/OpenRTX #280: rtxlink branch: openrtx_linux: Name of pty is not predictable

mbalmer, 2024-09-18:
openrtx_linux built from the rtxlink branch uses a PTY for communications and outputs the name to stdout.  This name, however, is of course not predictable and changes from invocation to invocation.

This is not very practical for daemon processes that talk to openrtx over rtxlink (e.g. trxd(8)), as there the device to control a transceiver has to be configured in a configuration file.

It would be desirable to have a deterministic name, e.g. sth that could be specified on the command line, or switch communication to a socket (Unix, IPv4, IPv6) instead of a pty.

silseva, 2024-09-18:
Yeah, is indeed not handy.
Do you have suggestions on how to create a named pseudo-tty interface?

Thanks!

mbalmer, 2024-09-18:
Not much you can do with PTYs, that is how they work. The best I could think of is to optionally specify a pathname on the commandline and then create a symlink to the pty.  So I could, e.g., in my homedirectory a file .openrtx.pty and hardcode this in my configuration.

Another solution, though a bit more of work, could be to switch to a socket.

silseva, 2024-10-16:
Solved with [a276c32](https://github.com/OpenRTX/OpenRTX/commit/a276c3259fc600321b6652069c045e6093fafd5f)

closed 2024-10-16 by silseva