OpenRTX/OpenRTX #308: CS7000-Plus branch has compile errors

rolf-tx, 2025-01-30:
CS7000-Plus has below compile errors:
meson compile -C build_arm openrtx_cs7000p

FAILED: openrtx_cs7000p.p/platform_drivers_NVM_W25Qx.c.o
arm-miosix-eabi-gcc -Iopenrtx_cs7000p.p -I. -I.. -I../platform/targets/CS7000-PLUS -I../openrtx/include -I../openrtx/include/rtx -I../openrtx/include/core -I../openrtx/include/calibration -I../openrtx/include/protocols -I../openrtx/include/fonts/adafruit -I../openrtx/include/fonts/symbols -I../platform/drivers/ADC -I../platform/drivers/NVM -I../platform/drivers/GPS -I../platform/drivers/SPI -I../platform/drivers/USB -I../platform/drivers/GPIO -I../platform/drivers/tones -I../platform/drivers/audio -I../platform/drivers/baseband -I../platform/drivers/backlight -I../platform/drivers/chSelector -I../lib/minmea/include -I../lib/qdec/include -I../platform/mcu/CMSIS/Include -I../platform/mcu/CMSIS/Device/ST/STM32H7xx/Include -I../platform/mcu/STM32H7xx -I../platform/mcu/STM32H7xx/drivers -I../lib/miosix-kernel/miosix -I../lib/miosix-kernel/miosix/config -I../lib/miosix-kernel/miosix/arch/common -I../lib/miosix-kernel/miosix/util -I../lib/miosix-kernel/miosix/arch/cortexM7 -Isubprojects/codec2 -I../subprojects/codec2 -I../subprojects/codec2/src -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -O0 -g -D_DEFAULT_SOURCE=1 -ffunction-sections -fdata-sections -Wall -Werror=return-type -g -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Os -DPLATFORM_CS7000P -Dtimegm=mktime -DFONT_UBUNTU_REGULAR '-DGIT_VERSION="v0.3.7-104-ga83ca437"' -DSTM32H743xx -DHSE_VALUE=25000000 -DDONT_USE_CMSIS_INIT -DCOMPILING_MIOSIX -D_POSIX_PRIORITY_SCHEDULING -D_ARCH_CORTEXM7 -MD -MQ openrtx_cs7000p.p/platform_drivers_NVM_W25Qx.c.o -MF openrtx_cs7000p.p/platform_drivers_NVM_W25Qx.c.o.d -o openrtx_cs7000p.p/platform_drivers_NVM_W25Qx.c.o -c ../platform/drivers/NVM/W25Qx.c
../platform/drivers/NVM/W25Qx.c: In function 'nvm_api_erase':
../platform/drivers/NVM/W25Qx.c:218:9: error: too few arguments to function 'enableWrite'
  218 |         enableWrite();
      |         ^~~~~~~~~~~
../platform/drivers/NVM/W25Qx.c:77:20: note: declared here
   77 | static inline void enableWrite(const struct W25QxCfg *cfg)
      |                    ^~~~~~~~~~~
../platform/drivers/NVM/W25Qx.c: In function 'W25Qx_writePage':
../platform/drivers/NVM/W25Qx.c:263:5: error: too few arguments to function 'enableWrite'
  263 |     enableWrite();
      |     ^~~~~~~~~~~
../platform/drivers/NVM/W25Qx.c:77:20: note: declared here
   77 | static inline void enableWrite(const struct W25QxCfg *cfg)
      |                    ^~~~~~~~~~~
../platform/drivers/NVM/W25Qx.c:272:10: error: 'offset' undeclared (first use in this function); did you mean 'off_t'?
  272 |         (offset >> 16) & 0xFF,  // Address high
      |          ^~~~~~
      |          off_t
../platform/drivers/NVM/W25Qx.c:272:10: note: each undeclared identifier is reported only once for each function it appears in
[19/56] Compiling C++ object openrtx_cs7000p.p/openrtx_src_rtx_OpMode_M17.cpp.o
ninja: build stopped: subcommand failed.

rolf-tx, 2025-01-30:
Fixed by copying platform/drivers/NVM/W25Qx.c from master branch

rolf-tx, 2025-01-30:
The  openrtx_cs7000p_bin does not appear to have the correct format for the m17 factory flash tool.
Any suggestions?

silseva, 2025-01-31:
Hi!
First of all a caveat: the CS7000-Plus branch, like all the branches not merged into master, has to be considered experimental. When making firmware images, you are proceeding at your own risk!
The firmware I sent to Connect Systems has been built from that code but it also has been tested on a device before sending it out.

This said:
1) I force-pushed the missing parts, now the branch compiles fine and the firmware is compatible with the bootloader
2) you need to set up the environment with `meson setup --cross-file cross_cm7.txt build_arm`

rolf-tx, 2025-01-31:
@silseva Totally understand the caveat emptor on experimental branch.

Thanks for pushing the missing parts branch now compiles ok.
However, I must be missing a step in generating and/or flashing procedure.
Which flash tool am I supposed to use?

1) I placed it in firmware update mode with sk1+orange->power on.
2) dfu-util and radio_tool do not see the cs7000p
3) I tried the factory tool flash on windows but it is complaining 'the file you just selected is not a host program type file'
4) The bin file is much smaller (184kB) than the one provided by connect systems (340kB)
 
This is how I built the image on Debian:
git clone --branch CS7000-Plus --single-branch --recursive https://github.com/OpenRTX/OpenRTX OpenRTX.CS7000-Plus
cd OpenRTX.CS7000-Plus/
git checkout CS7000-Plus
meson setup --cross-file cross_cm7.txt build_arm
meson compile -C build_arm openrtx_cs7000p_flash
ls -lh build_arm/openrtx_cs7000p_bin
-rwxrwxrwx 1 rolf rolf 182K Jan 31 10:04 build_arm/openrtx_cs7000p_bin

copy build_arm/openrtx_cs7000p_bin to windows as openrtx_cs7000p.bin

Thanks for your help!

silseva, 2025-02-03:
The file size is correct, the one I have on my computer is 195kB. To flash it on the radio I suggest you to follow the firmware upgrade procedure described on [CSI's website](https://www.connectsystems.com/amateur-software/CS7000_m17_plus_software.html).
You'll need the firmware image to be in DFU format, to generate it you just have to invoke the `openrtx_cs7000p_wrap` build target.

rolf-tx, 2025-02-04:
@silseva Thank you! Using build target openrtx_cs7000p_wrap was the missing piece.
I can now flash build_arm/openrtx_cs7000p.dfu using the 'bricked device' instructions from CSI.

closed 2025-02-04 by rolf-tx