OpenRTX/OpenRTX #321: Change battery reporting on TYT MD-UV380

majsterElektron, 2025-06-02:
Comparing to opengd77 reports 44%@7.2V on same radio as openrtx 0%
7.1V is 3.55 per cell, I think 0% should be something like 3.0 per cell

And radio sometimes freezes on "emergency use" screen and not reacting to any button. Looks like it does this with lower voltages.

silseva, 2025-06-04:
The battery voltage-to-charge curve we're using is probably a bit too much conservative. There is the possibility to use the 0% and 100% values from calibration data, but there is need for some reverse engineering first

majsterElektron, 2025-06-04:
That can be the issue, lipo have different discharge chart. Tyt uses liion.

I'd use something like this:

8.4=100%
8.2=100%
8.15=94%
8.1=94%
8.0=88%
7.9=83%
7.8=77%
7.7=72%
7.6=66%
7.5=61%
7.4=55%
7.3=50%
7.2=44%
7.1=38%
7.0=33%
6.9=27%
6.8=22%
6.7=16%
6.6=11%
6.5=5%
6.4=0%
6.3=0%

silseva, 2025-08-27:
Issue should now be solved by commits 0962b53 and 71c144d