OpenRTX/OpenRTX #329: GPS Enhancements to support new M17 GNSS spec

jancona, 2025-08-15:
Since you're working on the GPS driver... The new M17 GNSS spec has a couple of features that would benefit from driver support:

* There is a new Radius value, used to give an idea of the accuracy of the fix. It's deliberately coarse; the reported values are powers of two between 1 and 128 in meters. It's contemplated that this could be calculated from HDOP, so ideally that value would be available.
* There are now four validity bits for Lat/Lon, Altitude, Speed/Bearing, and Radius. On my CS7000, I notice that the GPS UI can show a 3D fix with a lat/lon of 0.0/0.0. If would be nice if the driver could indicate if the lat/lon and other values are valid.

Thanks.

silseva, 2025-08-15:
I think that the second point you mention is due to a bug in the current implementation, the new code should fix this issue and give valid data straight from the beginning

jancona, 2025-08-15:
I just tried your branch and don't see the zeroes. Thanks!

jancona, 2025-08-21:
I seem to be seeing HDOP values of zero on my CS7000, although I may be doing something wrong. In any case, is it safe to assume zero means "invalid"? Or maybe invalid should be 0xFFFF?

silseva, 2025-08-27:
GPS updates are now upstream!

closed 2025-08-27 by silseva

jancona, 2025-08-27:
That's great, thanks!