OpenRTX/OpenRTX #351: Support sending DTMF tones

turnrye, 2025-09-15:
As a ham radio operator utilizing FM mode with a repeater or hotspot,
I want to be able to use the number pad on my radio to send  DTMF tones
So that I can control the remote station

## Acceptance criteria
- The tones should only be sent while I'm holding PTT
- I should be able to enable and disable this feature via a setting
- This feature should only work when in FM mode
- I should hear the DTMF tones played on the speaker while I am transmitting them so that I'm aware of what is happening
- when I am transmitting DTMF tones, the mic should be disabled

## nonfunctional requirements
- this should be built in a scalable way so that the feature could be extended in the future to send defined DTMF sequences (e.g. for devices without a keypad, where an arrow based input is needed)

turnrye, 2025-09-15:
Some notes on tone generation, although in the context of beeps playing on the speaker and not being transmitted:

> The HR_Cx000 does not have any internal tone generator module, it only acts as an audio DAC with a FIFO of 64 signed 16-bit samples. So, beeps are generated taking samples from a sine table and sending them to the FIFO.
The same principle is used also in the MDx radios, with the only difference that there the audio signal is generated using filtered PWM

silseva, 2025-09-22:
Tracked in https://tasks.openrtx.org/project/openrtx/us/163