OpenRTX/OpenRTX #354: Project stance about AI contributions

mdiepart, 2025-09-18:
Let's open this issue by referencing what was discussed in PR #341.

The two AI policies that were linked in the PR are:
Cloud Hypervisor - https://www.phoronix.com/news/Cloud-Hypervisor-48
Gentoo - https://wiki.gentoo.org/wiki/Project:Council/AI_policy

IMO, the two main issues with AI contributions are
1) Copyright and risks of plagiarism or licenses infringement 
2) Risks of wasting maintainers time

There are, of course many more concerns for and against accepting AI contributions.

mdiepart, 2025-09-18:
Tagging @turnrye and @silseva :)

turnrye, 2025-09-19:
I think that it's important that we focus _strictly_ on generative AI, since historically developers have used AI in other areas without concern.

With generative AI, I do see where it can be harmful to the health of a project through flooding maintainers with questionable changes. And I agree with a lot of the other projects in this space that the nature of AI makes it more difficult to tell, because many of the typical signs of an incomplete contribution will be missing (generative AI is very convincing). This slop is the biggest fear of mine.

But, I also see where there are good applications for generative AI. For example, https://github.com/OpenRTX/OpenRTX/pull/345/commits/8fceb67e5c824ded1daac64f9d4d45ac0f002121 was generated using chatgpt. I think the linux foundation has taken a good stance on this. I think rather than blocking generative AI contributions, we should permit them given similar terms:

1. We get to change our mind, which means contributions using them need to be identifiable so that we can continue to assess the situation; this can be achieved by setting a standard of having a git commit message trailer, [like what was proposed for the linux kernel](https://lwn.net/ml/all/20250725175358.1989323-1-sashal@kernel.org/)
2. (I think we already all agree on this) We document the policy on the contributing guide, https://www.linuxfoundation.org/legal/generative-ai is a reasonable boilerplate
3. We establish the expectation and traceability for IP concerns by adopting the [DCO](https://developercertificate.org/), which probably means adding content like [this from k8s](https://cert-manager.io/docs/contributing/sign-off/) to the dev docs (contributing guide?) **and being sure to instruct AI _not_ to sign off [like what was proposed for the linux kernel](https://lwn.net/ml/all/20250725175358.1989323-1-sashal@kernel.org/)

lemielek, 2025-09-19:
Additional: https://www.phoronix.com/news/Mesa-Contributor-Guidelines ->>> https://gitlab.freedesktop.org/mesa/mesa/-/work_items/13736

turnrye, 2025-09-28:
https://communityblog.fedoraproject.org/council-policy-proposal-policy-on-ai-assisted-contributions/ was recently put for comment, and it is also consistent with the direction that we're considering in this thread.